GazzolaLab/PyElastica #289: Necessity of sanity checks in factory_function.py

saultshaker, 2023-08-07:
Hi PyElastica Team, 

I am creating 3D curved rods for some of my PyElastica simulations. In trying to instantiate them I throw an error in the `factory_function.py` file on line 41:

     `assert np.sqrt(np.dot(normal, normal)) > Tolerance.atol()`

To me, this code reads "check to make sure that the square root of the dot product of the vector that defines the normal director for the rod with itself is larger than the zero (~2e-12)"

What is this sanity check trying to ensure? Just that the normal vector is greater than zero?

armantekinalp, 2023-08-07:
@saultshaker yes purpose of that check is to make sure length of the normal vector is greater than zero.

bhosale2, 2023-08-08:
@saultshaker have your doubts been addressed? Were you able to fix the above error you were getting?

saultshaker, 2023-08-08:
@bhosale2 Yes, I was, thank you for checking in!

closed 2023-08-08 by saultshaker