GazzolaLab/PyElastica #290: Mesh demonstration/example

Ali-7800, 2023-08-08:
Create an example that uses the Mesh class to create a mesh surface, and use that to create an example based on the ContinuumSnakeTerrainCase example in [dev_snake_contact](https://github.com/Ali-7800/PyElastica/tree/dev_snake_contact)

Tasks:
- [ ] Derive a mesh_surface class from the surface class (similar to how sphere or cylinder is derived from rigid body), this should take the necessary attributes from the mesh class to be used for contact. 
- [ ] Derive a contact class from NoContact to create contact between a rod and a mesh_surface object, should be something similar to CustomInteractionSurface (multiple contacts) or CustomInteractionSurfaceGrid (multiple contacts) in the CustomFrictionSurface.py file.
- [ ] Create an example where the rod makes contact with a simple surface (not a plane)

Ali-7800, 2023-08-08:
@Rahul-JOON You can work on this

Rahul-JOON, 2023-08-08:
Sure thing!