GazzolaLab/PyElastica #295: Reproducing snake diffraction results from your Nat Comm Paper

AkashVardhan7, 2023-08-30:
Hello again, I am trying to develop a simulation of a worm passing through a pegged lattice, and started off with trying to replicate the results of snake diffraction and reflection and refraction reported in Fig 4 of your Nature communications paper.
I followed the instructions listed in the SI of your paper, and as a first pass tried to simulate a snake passing through 3 pegs located at (x=0, z = 0.75); (x=-0.23, z = 0.75) and (x=0.23, z = 0.75) where +z is the direction of snake travel.

[image]

The way I implemented this, is by modifying your continuumsnake.py example, and adding frictional forces corresponding to the patches, as shown in the following code snippet from the modified example.

# Add friction forces
    # Uniform friction with ground
    origin_plane = np.array([0.0, -base_radius, 0.0])
    normal_plane = normal
    slip_velocity_tol = 1e-8
    froude = 0.1
    mu = base_length / (period * period * np.abs(gravitational_acc) * froude)
    kinetic_mu_array = np.array(
        [mu, 1.5 * mu, 2.0 * mu]
    )  # [forward, backward, sideways]
    static_mu_array = np.zeros(kinetic_mu_array.shape)
    snake_diffraction_sim.add_forcing_to(shearable_rod).using(
        ea.AnisotropicFrictionalPlane,
        k=1.0,
        nu=1e-6,
        plane_origin=origin_plane,
        plane_normal=normal_plane,
        slip_velocity_tol=slip_velocity_tol,
        static_mu_array=static_mu_array,
        kinetic_mu_array=kinetic_mu_array,
    )

    # Add Patches at z = 0.75 m and several values of x
    # Patch 1 -> 0,-base_radius,0.75
    patch1_origin_plane = np.array([0.0,-base_radius,0.75])
    patch1_normal_plane = normal
    slip_velocity_tol = 1e-4
    patch1_froude = 0.1
    mu = base_length / (period * period * np.abs(gravitational_acc) * patch1_froude)
    p_factor1 = 30
    kinetic_mu_patch1_array = np.array (
        [mu*p_factor1, 1.5*mu*p_factor1, 2.0*mu*p_factor1]
    )
    static_mu_patch1_array = np.zeros(kinetic_mu_patch1_array.shape)
    snake_diffraction_sim.add_forcing_to(shearable_rod).using(
        ea.AnisotropicFrictionalPlane,
        k=1.0,
        nu=1e-6,
        plane_origin=patch1_origin_plane,
        plane_normal=patch1_normal_plane,
        slip_velocity_tol=slip_velocity_tol,
        static_mu_array=static_mu_patch1_array,
        kinetic_mu_array=kinetic_mu_patch1_array,
    )

    # Patch 2 -> 0.023,-base_radius,0.75
    patch2_origin_plane = np.array([0.023,-base_radius,0.75])
    patch2_normal_plane = normal
    slip_velocity_tol = 1e-4
    patch2_froude = 0.1
    mu = base_length / (period * period * np.abs(gravitational_acc) * patch2_froude)
    p_factor2 = 30
    kinetic_mu_patch2_array = np.array(
        [mu * p_factor2, 1.5 * mu * p_factor2, 2.0 * mu * p_factor2]
    )
    static_mu_patch2_array = np.zeros(kinetic_mu_patch1_array.shape)
    snake_diffraction_sim.add_forcing_to(shearable_rod).using(
        ea.AnisotropicFrictionalPlane,
        k=1.0,
        nu=1e-6,
        plane_origin=patch2_origin_plane,
        plane_normal=patch2_normal_plane,
        slip_velocity_tol=slip_velocity_tol,
        static_mu_array=static_mu_patch2_array,
        kinetic_mu_array=kinetic_mu_patch2_array,
    )

    # Patch 3 -> -0.023,-base_radius,0.75
    patch3_origin_plane = np.array([-0.023, -base_radius, 0.75])
    patch3_normal_plane = normal
    slip_velocity_tol = 1e-4
    patch3_froude = 0.1
    mu = base_length / (period * period * np.abs(gravitational_acc) * patch3_froude)
    p_factor3 = 30
    kinetic_mu_patch3_array = np.array(
        [mu * p_factor3, 1.5 * mu * p_factor3, 2.0 * mu * p_factor3]
    )
    static_mu_patch3_array = np.zeros(kinetic_mu_patch1_array.shape)
    snake_diffraction_sim.add_forcing_to(shearable_rod).using(
        ea.AnisotropicFrictionalPlane,
        k=1.0,
        nu=1e-6,
        plane_origin=patch3_origin_plane,
        plane_normal=patch3_normal_plane,
        slip_velocity_tol=slip_velocity_tol,
        static_mu_array=static_mu_patch3_array,
        kinetic_mu_array=kinetic_mu_patch3_array,
    )

I left the rest of the code unchanged, however while implementing this I had a question on defining the dimensions of the frictional patch which is supposed to represent a peg. The origin_plane which determines where the origin of the frictional substrate is going to be located only at that coordinate which we specify, however there is no way of specifying it's bounding dimensions such that when the snake is on any other part of the ground it should experience the standard frictional force, and only when it is in contact with the patch should it experience an increased friction. 

Clearly the simulation doesn't work as intended and you can see it in the gif I am attaching. What should I do to implement the frictional patches correctly. Thanks.

[image]

AkashVardhan7, 2023-08-31:
If I could find out how the ground element is created/referenced I would be able to partition into patches of higher friction based on the coordinates, but I can't find anything in your API which directly lets me extract coordinates of objects.

xzhan139, 2023-08-31:
Hello Akash, I can help with this. Yes, your understanding is correct - the "AnisotropicFrictionalPlane" doesn't have a boundary, so by adding many of them together will just duplicate friction force. 
The way to make local friction patches is to create a custom frictional plane. You can use the "AnisotropicFrictionalPlane" as a template, which should give you most of the basics you need. Then, the only change you want to make there, is to create a function to check the distance between your patch center and snake center. This sudo code may explain this:
---------
scale_friction  = 1
for every element/node on snake:
      compute distance between this element to patch center
      if distance < radius of the patch:
              scale_friction  = 30
---------
This way, you can have different friction for each segment on the snake, and only add the friction when the element is in the patch region. 
One more thing to add - in case you don't find the ground setup, it is in elastica/interaction.py
Please let me know if you have further question.

AkashVardhan7, 2023-08-31:
Hi thanks for your reply @xzhan139 . I had a follow up question, when I try to create the customfrictional plane I would still need a way to bound the dimensions of the plane, where are the commands to set the boundaries inside the customfrictional plane?

xzhan139, 2023-08-31:
Hi, let me explain this in another way. Your customfricitonal plane should start looking like a normal plane - no boundary and uniform friction everywhere. Then you just need to create a few high frictional spots on that plane. To do this, just scale the friction whenever snake elements are inside the region. So in the end, you just need one plane - your custom friction plane, which is an infinite plane, no bounds. Does this make sense?

AkashVardhan7, 2023-09-04:
I get the idea, I haven't gotten the code to work yet.
Basically what I did was , create a function like you suggested which checks if a given point is inside a patch defined by known coordinates. And it works for test cases,

import numpy as np

def is_point_inside_patch(point, patch_vertices):
    """
    Check if a 3D point is inside a patch defined by four vertices.

    Args:
        point (numpy.ndarray): 3D coordinates of the point.
        patch_vertices (list of numpy.ndarray): List of four 3D coordinates defining the patch.

    Returns:
        bool: True if the point is inside the patch, False otherwise.
    """
    # Calculate the normal vectors of the four triangles formed by the point and patch vertices
    normals = []
    for i in range(4):
        vertex1 = patch_vertices[i]
        vertex2 = patch_vertices[(i + 1) % 4]
        normal = np.cross(vertex2 - vertex1, point - vertex1)
        normals.append(normal)

    # Check if the point is on the same side of all the patch's triangles (using dot product)
    for i in range(4):
        if np.dot(normals[i], normals[(i + 1) % 4]) < 0:
            return False

    return True

# Test cases
if __name__ == "__main__":
    # Define patch vertices (four known points in 3D space)
    patch_vertices = [
        np.array([0.0, 0.0, 0.0]),
        np.array([0.5, 0.0, 0.0]),
        np.array([0.5, 0.5, 0.0]),
        np.array([0.0, 0.5, 0.0]),
    ]

    # Generate 10 random points
    random_points = np.random.uniform(size=(10, 3))

    # Check if each random point is inside the patch and print the result
    for i, point in enumerate(random_points):
        is_inside = is_point_inside_patch(point, patch_vertices)
        print(f"Point {i + 1}: {point} is inside the patch: {is_inside}")
        
        
        
        
        
        Now when I do the same thing in the main sim, snake_diffraction.py. Which I am copy pasting... I am running into an error.
        
        
        Here's the code.
        import os
import numpy as np
import elastica as ea

def is_point_inside_patch(point, patch_vertices):
    """
    Check if a 3D point is inside a patch defined by four vertices.

    Args:
        point (numpy.ndarray): 3D coordinates of the point.
        patch_vertices (list of numpy.ndarray): List of four 3D coordinates defining the patch.

    Returns:
[... 331 lines cut ...]
            t_coeff_optimized, PLOT_FIGURE, SAVE_FIGURE, SAVE_VIDEO, SAVE_RESULTS
        )

        print("average forward velocity:", avg_forward)
        print("average forward lateral:", avg_lateral)

And here's the error...

Traceback (most recent call last):
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\pydevconsole.py", line 364, in runcode
    coro = func()
  File "<input>", line 1, in <module>
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\_pydev_bundle\pydev_umd.py", line 198, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "C:\Users\avardhan7\AppData\Local\JetBrains\PyCharm 2023.1.2\plugins\python\helpers\pydev\_pydev_imps\_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 351, in <module>
    [avg_forward, avg_lateral, pp_list] = run_snake(
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 143, in run_snake
    if is_point_inside_patch(node_position, patch_vertices):
  File "C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 22, in is_point_inside_patch
    normal = np.cross(vertex2 - vertex1, point - vertex1)
ValueError: operands could not be broadcast together with shapes (51,) (3,) 

I haven't had time to debug this...you can choose to keep this thread open or close it, and I will post with the status once I am able to resolve the issue. Thanks.

AkashVardhan7, 2023-09-10:
Finally got around to getting it to work, there was a minor error with the loop syntax.
Here's a video. Thanks for your help. 
[image]

armantekinalp, 2023-09-10:
Seems like this issue has been resolved, I am closing it.

closed 2023-09-10 by armantekinalp