GazzolaLab/PyElastica #299: Failure for Helix under Gravity Case

QuantuMope, 2023-09-14:
I am trying to simulate a helical rod deforming under gravity using PyElastica but I am running into instability no matter the time step I use. For this case, I expect the tip deflections to look something like this:

[image]

Above are results using Discrete Elastic Rods (DER).
But when I use PyElastica to simulate the results, I end up getting these results:

[image]

Above are results using a timestep size of 1e-5. Even when I use 1e-6, the same instability happens.
Would you happen to know why this instability happens?
I have attached the script I used to run the example along with all necessary files. Note that the directors of the rod are the same ones used in my DER example.

**To Reproduce**
```py
import numpy as np
import elastica as ea
from elastica._linalg import _batch_norm

class HelixGravitySimulator(ea.BaseSystemCollection, ea.Constraints, ea.Forcing, ea.CallBacks):
    pass

class HelixCallBack(ea.CallBackBaseClass):
    def __init__(self, step_skip: int, callback_params: dict):
        ea.CallBackBaseClass.__init__(self)
        self.every = step_skip
        self.callback_params = callback_params

    def make_callback(self, system, time, current_step: int):
        if current_step % self.every == 0:
            self.callback_params["time"].append(time)
            self.callback_params["position"].append(
                system.position_collection.copy()
            )
            self.callback_params["deflection"].append(
                system.position_collection[2, -1].copy()
            )
            return

positions = np.loadtxt("helix_configuration.txt")
positions = positions.T

n_elem = positions.shape[-1] - 1

tangents = positions[..., 1:] - positions[..., :-1]
tangents /= _batch_norm(tangents)

m1 = np.loadtxt("material_frames_m1.txt")
m2 = np.loadtxt("material_frames_m2.txt")
# Need to redo this, even though they should already be unit vectors
m1 /= np.linalg.norm(m1, axis=1).reshape((-1, 1))
m2 /= np.linalg.norm(m2, axis=1).reshape((-1, 1))

m1 = m1.T
m2 = m2.T

directors = np.zeros((3, 3, n_elem))
directors[0, :, :] = m1
directors[1, :, :] = m2
directors[2, :, :] = tangents

# Should already be orthonormal, but PyElastica is saying not accurate enough
# Probably from float number precision loss during saving
for i in range(n_elem):
    directors[1, :, i] = np.cross(directors[2, :, i], directors[0, :, i])
    directors[1, :, i] /= np.linalg.norm(directors[1, :, i])

shearable_rod = ea.CosseratRod.straight_rod(n_elem,
                                            positions[:, 0],
                                            tangents[:, 0],
                                            [1, 0, 0],
                                            0.5445,
                                            0.005,
                                            1273.52,
                                            youngs_modulus=1e7,
                                            position=positions,
                                            directors=directors)

shearable_rod.rest_kappa[:] = shearable_rod.kappa[:]
shearable_rod.rest_sigma[:] = shearable_rod.sigma[:]

helix_sim = HelixGravitySimulator()
helix_sim.append(shearable_rod)
helix_sim.constrain(shearable_rod).using(ea.OneEndFixedBC,
                                         constrained_position_idx=(0,),
                                         constrained_director_idx=(0,))

# Add gravitational forces
gravitational_acc = -9.80665
helix_sim.add_forcing_to(shearable_rod).using(
    ea.GravityForces, acc_gravity=np.array([0.0, 0.0, gravitational_acc])
)

dt = 1e-5
final_time = 10
total_steps = int(final_time / dt)
rendering_fps = 120
step_skips = int(1.0 / (rendering_fps * dt))

recorded_history = ea.defaultdict(list)
helix_sim.collect_diagnostics(shearable_rod).using(HelixCallBack, step_skips, callback_params=recorded_history)

helix_sim.finalize()

timestepper = ea.PositionVerlet()

ea.integrate(timestepper, helix_sim, final_time, total_steps)

import matplotlib.pyplot as plt

time = np.asarray(recorded_history['time'])
results = np.asarray(recorded_history['position'])

fig = plt.figure(figsize=(20, 5))
plt.plot(time, results[:, -1, -1])
plt.savefig("helix_dt={}.png".format(dt))
#
# fig = plt.figure()
# for i in range(time.shape[0]):
#     ax = fig.add_subplot(projection='3d')
#     ax.set_xlim(-0.1, 0.1)
#     ax.set_ylim(-0.1, 0.1)
#     ax.set_zlim(-0.2, 0.2)
#
#     ax.plot(results[i, 0, :], results[i, 1, :], results[i, 2, :])
#
#     plt.savefig("helix_frames/frame{}.png".format(i))
#     plt.clf()

```
[helix_configuration.txt](https://github.com/GazzolaLab/PyElastica/files/12612287/helix_configuration.txt)
[material_frames_m1.txt](https://github.com/GazzolaLab/PyElastica/files/12612285/material_frames_m1.txt)
[material_frames_m2.txt](https://github.com/GazzolaLab/PyElastica/files/12612286/material_frames_m2.txt)

Ali-7800, 2023-09-20:
Hi @QuantuMope, 
I have been looking at you're code and seems to be running fine, I think what is causing the numerical issues you have is the first part of your rod where the straight part unnaturally joins the rest of the helix. When I removed the first six elements I got a better result at dt = 1e-5.
[image]
Let me know if this fixes your issue/

QuantuMope, 2023-09-28:
I see. I wonder why that part causes issues when it doesn't for DER. Regardless, thank you for looking into this!

closed 2023-09-28 by QuantuMope