GazzolaLab/PyElastica #300: Can PyElastica support GPU acceleration?

Junang-Wang, 2023-09-22:
Hi, all. I am a beginner of PyElastica. Are PyElastica supporting GPU acceleration, e.g.  using cuda or taichi? Cases like someone might want to simulate independent and multiple rods? Or simulating one rod in muliple force configurations?

Junang-Wang, 2023-09-25:
Hi @armantekinalp,
Thanks for your reply. I have another question when I simulate a pendulum. The simulation is numerical unstable, i.e. elements of the rod explode away  after a few seconds. Do you know how to fix it?  I attach the files below.

Thanks u for advance. 

https://github.com/GazzolaLab/PyElastica/assets/89964840/e1992fb9-5446-4ebe-bc42-6edde16c6cf0
[Pendulum with string.ipynb.zip](https://github.com/GazzolaLab/PyElastica/files/12712235/Pendulum.with.string.ipynb.zip)

skim0119, 2023-10-12:
@Junang-Wang If you are still having this issue, please make a separate issue for this specific problem. It is easy to slip our attention if you post a question on unrelated thread.

Judging from the video, it appears that the moduli are not set correctly. I would recommend checking the value for moduli and damping constants.

I'm closing this issue as we do not have an immediate plan to include GPU acceleration.

closed 2023-10-12 by skim0119