GazzolaLab/PyElastica #304: Do PyElastica support to simulate a rod containing two segments?

Junang-Wang, 2023-10-12:
Hi, 

I currently working on a project where we want to simulate a rod made by two different materials. For example, in the following figure, one segment of rod may have different stiffness or Young's mudulus comparing to the other segment. Is there a way to simulate this case by PyElastica? 
[image]

Thanks for advance.

Ali-7800, 2023-10-13:
Hi @Junang-Wang,

Yes it is possible, you have to create two rods with your desired Young's moduli/radii and join them together with one of the joints in joint.py depending on your application. If you want them to act like one rod, your best bet is to use the FixedJoint class.
To see how to use the joint classes you can take a look at the `examples/JointCases` folder.

Let me know if you need more help!

Junang-Wang, 2023-10-16:
Hi Ali, Thanks a lot for your help! You save my day.

Ali-7800, 2023-10-16:
Hi @Junang-Wang, you're welcome! I will close this issue now, if you need any other help you can create a new one and we'll help you!

closed 2023-10-16 by Ali-7800