GazzolaLab/PyElastica #305: Question about base length

SHITIANYU-hue, 2023-10-12:
Hello, this is how I created this rod, my base length is 10, however, when i run the simulation, it much more than the length that i set:
[image]

density = 7500000 
nu = 0.1 #damping constant
E = 1.2e9 
poisson_ratio = 10
shear_modulus = E / (poisson_ratio + 1.0)

start = np.array([0.0,0.0,1.1])
direction = np.array([0.0, 1.0, 0.0])
normal = np.array([1.0, 0.0, 0.0])
base_length = 10
base_radius = 0.05
base_area = np.pi * base_radius**2

origin_force = np.array([0.0, 0.0, 0.0])
end_force = np.array([0.0, 1, 1])
ramp_up_time = 0.1

shearable_rod = CosseratRod.straight_rod(
    n_elem,
    start,
    direction,
    normal,
    base_length,
    base_radius,
    density,
    youngs_modulus=E,
    shear_modulus = shear_modulus,
)

Ali-7800, 2023-10-13:
Hi @SHITIANYU-hue,

Can you provide us with more detail on the rest of the simulation to figure out exactly what is wrong here, like what forces are you applying, how are you plotting you results, and what are you trying to achieve? 

The screenshot you posted looks like it is showing the simulation at 8.6s, so depending on what your doing with you simulation the length of your rod could change from its initial length of 10.

SHITIANYU-hue, 2023-10-16:
Hello @Ali-7800 , thanks for your reply! i applied the force like this: %matplotlib inline
time = 0
class PendulumSimulator(BaseSystemCollection, Constraints, Forcing, Damping):
    pass 

Pendulum_Sim = PendulumSimulator()

# Create Rod 
n_elem = 50

density = 75000
nu = 0.1 #damping constant
E = 1.2e9 
poisson_ratio = 10
shear_modulus = E / (poisson_ratio + 1.0)

start = np.array([0.0,0.0,1.1])
direction = np.array([0.0, 1.0, 0.0])
normal = np.array([1.0, 0.0, 0.0])
base_length = 1
base_radius = 1/83
base_area = np.pi * base_radius**2

origin_force = np.array([0.0, 0.0, 0.0])
end_force = np.array([0.0, 3, 3])
ramp_up_time = 0.1

shearable_rod = CosseratRod.straight_rod(
    n_elem,
    start,
    direction,
    normal,
    base_length,
    base_radius,
    density,
    youngs_modulus=E,
    shear_modulus = shear_modulus,
)

dl = base_length / n_elem 
dt = 0.0002 * dl 
Pendulum_Sim.append(shearable_rod)
Pendulum_Sim.dampen(shearable_rod).using(
    AnalyticalLinearDamper,
    damping_constant = nu,
    time_step = dt,
)

Pendulum_Sim.constrain(shearable_rod).using(
    OneEndFixedRod,constrained_position_idx = (0,), constrained_director_idx= (0,)
)

print("One end of the rod is now fixed in place")

Pendulum_Sim.add_forcing_to(shearable_rod).using(
    GravityForces, acc_gravity = np.array([0.0,0.0,-9.8]) 
)

Pendulum_Sim.add_forcing_to(shearable_rod).using(
    EndpointForces, origin_force, end_force, ramp_up_time = ramp_up_time
)

print("Forces added to the rod")

Pendulum_Sim.finalize()
print("System finalized")
#integrate(timestepper, Pendulum_Sim, final_time, total_steps)
def run_and_update_plot(simulator, dt, start_time, stop_time, ax):
    timestepper = PositionVerlet()
    do_step, stages_and_updates = extend_stepper_interface(timestepper, simulator)

    n_steps = int((stop_time - start_time)/ dt)
    time = start_time
    for i in range(n_steps):
        time = do_step(timestepper, stages_and_updates, simulator, time, dt)
    plot_pendulum_dynamic(shearable_rod, time, ax)
    return time 

def plot_pendulum_dynamic(rod1, time, ax):
    ax.clear()
    ax.grid(which="major", color="grey", linestyle="-", linewidth=0.25)
    ax.plot(
        rod1.position_collection[1,:],
        rod1.position_collection[2,:],
        "b-",
        label = "shearable rod",
    )

    ax.legend(prop={"size":12}, loc="lower left")
    ax.set_ylabel("Y Position (1mm)", fontsize=12)
    ax.set_xlabel("X Position (1mm)", fontsize=12)
    ax.set_title(f"Simulation Time: {time:.2f} seconds")
    ax.set_ylim([-2,2])
    ax.set_xlim([-2,2])

Based on my understanding, the rod should not be Infinitely extended, but I observed in the simulation, it is Infinitely extended, may I know if the simulation collapsed? But if i changed another set of density and E , the rod will not be Infinitely extended, 

[image]

Ali-7800, 2023-10-16:
Hi @SHITIANYU-hue, 
What timestep are you using when you call run_and_update_plot? when I use dt = 0.0002 * dl, I get:
[image]
Here's a video of the whole simulation:

https://github.com/GazzolaLab/PyElastica/assets/47090295/f32538f0-e7ee-4b9a-b362-748a25d90a6d

SHITIANYU-hue, 2023-10-17:
Hi @Ali-7800 , i am really sorry I used another set of parameters, and thanks for running experiment, i add my full code below: 

```
import numpy as np
import matplotlib.pyplot as plt
from IPython import display
# Import modules
from elastica.modules import BaseSystemCollection, Constraints, Forcing, Damping

# Import Cosserat Rod Class
from elastica.rod.cosserat_rod import CosseratRod

# Import Damping Class
from elastica.dissipation import AnalyticalLinearDamper

# Import Boundary Condition Classes
from elastica.boundary_conditions import OneEndFixedRod, FreeRod
from elastica.external_forces import EndpointForces 
from elastica.external_forces import GravityForces

# Import Timestepping Functions
from elastica.timestepper.symplectic_steppers import PositionVerlet 
from elastica.timestepper import extend_stepper_interface
from elastica.timestepper import integrate 
%matplotlib inline
time = 0
class PendulumSimulator(BaseSystemCollection, Constraints, Forcing, Damping):
    pass 

Pendulum_Sim = PendulumSimulator()

# Create Rod 
n_elem = 50

density = 7500000
nu = 0.1 #damping constant
E = 1.2e9 
poisson_ratio = 10
shear_modulus = E / (poisson_ratio + 1.0)

start = np.array([0.0,0.0,1.1])
direction = np.array([0.0, 1.0, 0.0])
normal = np.array([1.0, 0.0, 0.0])
base_length = 10
base_radius = 1/83
base_area = np.pi * base_radius**2

origin_force = np.array([0.0, 0.0, 0.0])
end_force = np.array([0.0, 3, 3])
ramp_up_time = 0.1

shearable_rod = CosseratRod.straight_rod(
    n_elem,
    start,
    direction,
    normal,
    base_length,
    base_radius,
    density,
    youngs_modulus=E,
    shear_modulus = shear_modulus,
)

dl = base_length / n_elem 
dt = 0.0002 * dl 
Pendulum_Sim.append(shearable_rod)
Pendulum_Sim.dampen(shearable_rod).using(
    AnalyticalLinearDamper,
    damping_constant = nu,
    time_step = dt,
)

Pendulum_Sim.constrain(shearable_rod).using(
    OneEndFixedRod,constrained_position_idx = (0,), constrained_director_idx= (0,)
)

print("One end of the rod is now fixed in place")

Pendulum_Sim.add_forcing_to(shearable_rod).using(
    GravityForces, acc_gravity = np.array([0.0,0.0,-9.8]) 
)

Pendulum_Sim.add_forcing_to(shearable_rod).using(
    EndpointForces, origin_force, end_force, ramp_up_time = ramp_up_time
)

print("Forces added to the rod")

Pendulum_Sim.finalize()
print("System finalized")
#integrate(timestepper, Pendulum_Sim, final_time, total_steps)
def run_and_update_plot(simulator, dt, start_time, stop_time, ax):
    timestepper = PositionVerlet()
    do_step, stages_and_updates = extend_stepper_interface(timestepper, simulator)

    n_steps = int((stop_time - start_time)/ dt)
    time = start_time
    for i in range(n_steps):
        time = do_step(timestepper, stages_and_updates, simulator, time, dt)
    plot_pendulum_dynamic(shearable_rod, time, ax)
    return time 

def plot_pendulum_dynamic(rod1, time, ax):
    ax.clear()
    ax.grid(which="major", color="grey", linestyle="-", linewidth=0.25)
    ax.plot(
        rod1.position_collection[1,:],
        rod1.position_collection[2,:],
        "b-",
        label = "shearable rod",
    )

    ax.legend(prop={"size":12}, loc="lower left")
    ax.set_ylabel("Y Position (1mm)", fontsize=12)
    ax.set_xlabel("X Position (1mm)", fontsize=12)
    ax.set_title(f"Simulation Time: {time:.2f} seconds")
    ax.set_ylim([-40,40])
    ax.set_xlim([-10,10])
%matplotlib inline
evolve_for_time = 15.0
update_interval = 1.0e-1 

import cv2,os
# update the plot every 0.1 second
fig = plt.figure(figsize=(5,4), frameon=True, dpi=150)
ax = fig.add_subplot(111)

plt.savefig('Pendulum.jpg')
video_name = 'Pendulum.mp4'
frame = cv2.imread('Pendulum.jpg')
height, width, layers = frame.shape
fourcc = cv2.VideoWriter_fourcc('M','P','4','V') # 解码器
video = cv2.VideoWriter(video_name,fourcc,1,(width,height))
first_interval_time = update_interval + time 
last_interval_time = time + evolve_for_time
for stop_time in np.arange(
    first_interval_time, last_interval_time+dt, update_interval
):
    time = run_and_update_plot(Pendulum_Sim, dt, time, stop_time, ax)
    display.clear_output(wait=True)
    image = plt.gcf()
    display.display(image)
    plt.savefig('Pendulum.jpg')
    video.write(cv2.imread("Pendulum.jpg"))
plt.close()
cv2.destroyAllWindows()
video.release()

```

This is my simulation: 

https://github.com/GazzolaLab/PyElastica/assets/38762389/f6ba9e4c-945e-4e36-862c-39765c1f2a78

Ali-7800, 2023-10-18:
Hi @SHITIANYU-hue,

May I ask what exactly are you trying to simulate here, so I can better understand how to help you? Because depending on the units/dimensions you use you will get a different simulation and you might need to change the timestepping accordingly.

Also, is there a particular reason that you are not using the callback module for the plotting? I think it would be easier to do something similar to the ContinuumSnakeCase example where we do the entire simulation then plot the video instead of what you are doing.

SHITIANYU-hue, 2023-10-19:
Thanks for your reply! i understand i need to change the time-stepping accordingly, i will also try to use callback module for the plotting. I am working on the project similar to this issue: https://github.com/GazzolaLab/PyElastica/issues/310 

I think in another repository build upon your work also discussed about the numerical simulation difficulties in Pyelasitca: https://github.com/alantes/RL-for-MSRs/issues/2 .

bhosale2, 2024-02-01:
@SHITIANYU-hue has the issue been resolved? Or else we shall close the issue due to inactivity in a few days.

SHITIANYU-hue, 2024-02-01:
Yes, thanks!