GazzolaLab/PyElastica #306: Replicating side-winding results from your Nat-comm snake paper

AkashVardhan7, 2023-10-12:
Hello again., I have been trying to replicate some of the side-winding results from your paper, and added a separate class to compute the torques from the lifting wave in the external_forces.py file like the MuscleTorques class, and then in the main simulation file, I add the forcing from the lifting wave using the liftingTorques class.. like it's done for the lateral wave with the MuscleTorques class..however it seems I will have to re-build/compile elastica again as I get the following error message AttributeError: module 'elastica' has no attribute 'LiftingTorques'. I am pasting the codes for these files...kindly take a peek and let me know if I am on the right track before I re-compile elastica to reflect the modification with the LiftingTorques.

here's the modified external_forces.py file with the modified Lifting wave
__doc__ = """ Numba implementation module for boundary condition implementations that apply
external forces to the system."""

import numpy as np
from elastica._linalg import _batch_matvec
from elastica.typing import SystemType, RodType
from elastica.utils import _bspline

from numba import njit
from elastica._linalg import _batch_product_i_k_to_ik

class NoForces:
    """
    This is the base class for external forcing boundary conditions applied to rod-like objects.

    Notes
    -----
    Every new external forcing class must be derived
    from NoForces class.

    """

    def __init__(self):
        """
        NoForces class does not need any input parameters.
        """
        pass

    def apply_forces(self, system: SystemType, time: np.float64 = 0.0):
        """Apply forces to a rod-like object.

        In NoForces class, this routine simply passes.

        Parameters
        ----------
        system : SystemType
            Rod or rigid-body object
        time : float
            The time of simulation.

        Returns
        -------

        """
        pass

    def apply_torques(self, system: SystemType, time: np.float64 = 0.0):
        """Apply torques to a rod-like object.

        In NoForces class, this routine simply passes.

        Parameters
        ----------
        system : SystemType
            Rod or rigid-body object
        time : float
            The time of simulation.

        Returns
        -------

        """
        pass

class GravityForces(NoForces):
    """
    This class applies a constant gravitational force to the entire rod.
[... 997 lines cut ...]
        filename_data = "optimized_coefficients.txt"
        if SAVE_OPTIMIZED_COEFFICIENTS:
            assert filename_data != "", "provide a file name for coefficients"
            np.savetxt(filename_data, optimized_spline_coefficients, delimiter=",")

    else:
        # Add muscle forces on the rod
        if os.path.exists("optimized_coefficients.txt"):
            t_coeff_optimized = np.genfromtxt(
                "optimized_coefficients.txt", delimiter=","
            )
        else:
            wave_length = 1.0
            t_coeff_optimized = np.array(
                [3.4e-3, 3.3e-3, 4.2e-3, 2.6e-3, 3.6e-3, 3.5e-3]
            )
            t_coeff_optimized = np.hstack((t_coeff_optimized, wave_length))

        # run the simulation
        [avg_forward, avg_lateral, pp_list] = run_snake(
            t_coeff_optimized, PLOT_FIGURE, SAVE_FIGURE, SAVE_VIDEO, SAVE_RESULTS
        )

        print("average forward velocity:", avg_forward)
        print("average forward lateral:", avg_lateral)

Ali-7800, 2023-10-13:
Hi @AkashVardhan7 ,

In elastica we have an `__init__.py` file which is used for importing different functions/classes, if you take a look at it you can see that we are importing all the classes in `external_forces.py` like this:

`from elastica.external_forces import (
    NoForces,
    EndpointForces,
    GravityForces,
    UniformForces,
    UniformTorques,
    MuscleTorques,
    EndpointForcesSinusoidal,
)`

if you want use the class you just created the way you are doing it now, you have to add it there too. Alternatively, you can put your class in the simulation file with all the necessary imports and use it without the `ea.`. One last thing, in order for this class to work it must also have the method `apply_torques` so it can be called in the backend, right now you have it called `apply__lifting_torques` which would not work.

Let me know if that helps!

AkashVardhan7, 2023-10-13:
I modified the __init__ file to have the LiftingTorques in from elastica.external_forces import(),. and modified the def apply_torques(self, rod: RodType, time: np.float64 = 0.0): as you suggested in the external_forces.py... the error I get now has to di with the forcing.py... Here are the modified code snippets and the error message...

from elastica.external_forces import (
    NoForces,
    EndpointForces,
    GravityForces,
    UniformForces,
    UniformTorques,
    MuscleTorques,
    LiftingTorques,
    EndpointForcesSinusoidal,
)

def apply_torques(self, rod: RodType, time: np.float64 = 0.0):
    self.compute_lifting_torques(
        time,
        self.my_spline,
        self.s,
        self.angular_frequency,
        self.wave_number,
        self.Amplitude,
        self.lamBDA,
        self.phase_shift,
        self.ramp_up_time,
        self.direction,
        rod.director_collection,
        rod.external_torques,
    )
def apply_torques(self, rod: RodType, time: np.float64 = 0.0):
    self.compute_lifting_torques(
        time,
        self.my_spline,
        self.s,
        self.angular_frequency,
        self.wave_number,
        self.Amplitude,
        self.lamBDA,
        self.phase_shift,
        self.ramp_up_time,
        self.direction,
        rod.director_collection,
        rod.external_torques,
    )

/Users/avardhan7/Documents/OPEN_AI/pyElastica/snake_diffraction/snake_diffraction.py
Traceback (most recent call last):
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\forcing.py", line 171, in __call__
    return self._forcing_cls(*self._args, **self._kwargs)
TypeError: NoForces.__init__() got an unexpected keyword argument 'base_length'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 379, in <module>
    [avg_forward, avg_lateral, pp_list] = run_snake(
  File "c:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 280, in run_snake        
    snake_diffraction_sim.finalize()
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\base_system.py", line 163, in finalize    
    finalize()
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\forcing.py", line 59, in _finalize_forcing    self._ext_forces_torques[:] = [
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\forcing.py", line 60, in <listcomp>       
    (ext_force_torque.id(), ext_force_torque())
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\forcing.py", line 173, in __call__        
    raise TypeError(
TypeError: Unable to construct forcing class.\nDid you provide all necessary force properties?
PS C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction>

Ali-7800, 2023-10-13:
It looks like the error is that you are passing base_length to a forcing class that does not take base_length, did you make any other changes to your code other than I told you about?

AkashVardhan7, 2023-10-16:
I didn't. The only changes I made are the following.
1. Put the class LiftingTorques inside external_forces.py {when I tried running it from the main simulation after appropriately pasting all the necessary imports and also the class NoForces.py... I got the following message... Assertion Error: <class '__main__.LiftingTorques'> is not a valid forcing. Did you forget to derive from NoForces?....so went back to running with having the class in external_forces.py}.

I am pasting this class again., which I modified from the class MuscleTorques...base_length is passed as an input to def__init__ and then in the main simulation it is called by the LiftingTorques class by passing the same base_length value that is passed to the class MuscleTorques.
class LiftingTorques(NoForces):
  """
  This class applies Lifting torques along the body. The applied muscle torques are treated
  as applied external forces. This class can apply
  muscle torques as a traveling wave with a beta spline or only
  as a traveling wave. For implementation details refer to Gazzola et. al.
  RSoS. (2018).

    Attributes
    ----------
    direction: numpy.ndarray
        2D (dim, 1) array containing data with 'float' type. Muscle torque direction.
    angular_frequency: float
        Angular frequency of traveling wave.
    wave_number: float
        Wave number of traveling wave.
    phase_shift: float
        Phase shift of traveling wave.
    ramp_up_time: float
        Applied muscle torques are ramped up until ramp up time.
    my_spline: numpy.ndarray
        1D (blocksize) array containing data with 'float' type. Generated spline.

  """

def __init__(
        self,
        base_length,
        b_coeff,
        period,
        wave_number,
        Amplitude,
        lamBDA,
        phase_shift,
        direction,
        rest_lengths,
        ramp_up_time,
        with_spline=False,
):
    """

    Parameters
    ----------
    base_length: float
        Rest length of the rod-like object.
    b_coeff: nump.ndarray
        1D array containing data with 'float' type.
        Beta coefficients for beta-spline.
    period: float
        Period of traveling wave.
    wave_number: float
        Wave number of traveling wave.
    Amplitude : flot
    lamBDA : float
    phase_shift: float
        Phase shift of traveling wave.
    direction: numpy.ndarray
       1D (dim) array containing data with 'float' type. Muscle torque direction.
    ramp_up_time: float
        Applied muscle torques are ramped up until ramp up time.
    with_spline: boolean
        Option to use beta-spline.

    """
    super(LiftingTorques, self).__init__()

[... 105 lines cut ...]
      period=period,
      wave_number=2.0 * np.pi / (wave_length),
      Amplitude=0.7,
      lamBDA=1,
      phase_shift=0.25,
      rest_lengths=snake_body.rest_lengths,
      ramp_up_time=period,
      direction=direction,
      with_spline=True,
    )
    
    
    and here' the addition of the class to the __init__.py file..
    from elastica.external_forces import (
    NoForces,
    EndpointForces,
    GravityForces,
    UniformForces,
    UniformTorques,
    MuscleTorques,
    LiftingTorques,
    EndpointForcesSinusoidal,
)

Besides, these I made no changes to any of the files..

Ali-7800, 2023-10-17:
I think the problem is with the indentation of what is inside of your class, make sure that you indent everything in the class correctly. Let me know how it goes.

AkashVardhan7, 2023-10-17:
There was an indentation error, which I addressed, but now I am getting the following message,

Traceback (most recent call last):
  File "c:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 379, in <module>
    [avg_forward, avg_lateral, pp_list] = run_snake(
  File "c:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction\snake_diffraction.py", line 283, in run_snake
    ea.integrate(timestepper, snake_diffraction_sim, final_time, total_steps)
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\timestepper\__init__.py", line 106, in integrate
    time = do_step(StatefulStepper, stages_and_updates, System, time, dt)
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\timestepper\symplectic_steppers.py", line 90, in do_step
    SystemCollection.synchronize(time)
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\base_system.py", line 175, in synchronize
    feature(time)
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\modules\forcing.py", line 86, in _call_ext_forces_torques
    ext_force_torque.apply_torques(self._systems[sys_id], time, *args, **kwargs)
  File "C:\Users\avardhan7\.conda\envs\pyELastica\lib\site-packages\elastica\external_forces.py", line 483, in apply_torques
    self.compute_lifting_torques(
TypeError: LiftingTorques.compute_lifting_torques() takes 12 positional arguments but 13 were given
PS C:\Users\avardhan7\Documents\OPEN_AI\pyElastica\snake_diffraction> 

This is my current class for applying LiftingTorques

class LiftingTorques(NoForces):
    """
    This class applies Lifting torques along the body. The applied muscle torques are treated
    as applied external forces. This class can apply muscle torques as a traveling wave with a beta spline or only
    as a traveling wave. For implementation details refer to Gazzola et. al. RSoS. (2018).
    """

    def __init__(
        self,
        base_length,
        b_coeff,
        period,
        wave_number,
        Amplitude,
        lamBDA,
        phase_shift,
        direction,
        rest_lengths,
        ramp_up_time,
        with_spline=False
    ):
        super(LiftingTorques, self).__init__()

        self.direction = direction  # Direction torque applied
        self.Amplitude = Amplitude
        self.lamBDA = lamBDA
        self.angular_frequency = 2.0 * np.pi / period
        self.wave_number = wave_number
        self.phase_shift = phase_shift

        assert ramp_up_time > 0.0
        self.ramp_up_time = ramp_up_time

        # s is the position of nodes on the rod, we go from node=1 to node=nelem-1, because there is no
        # torques applied by first and last node on elements. Reason is that we cannot apply torque in an
        # infinitesimal segment at the beginning and end of rod, because there is no additional element
        # (at element=-1 or element=n_elem+1) to provide internal torques to cancel out an external
        # torque. This coupled with the requirement that the sum of all muscle torques has
        # to be zero results in this condition.
        self.s = np.cumsum(rest_lengths)
        self.s /= self.s[-1]

        if with_spline:
            assert b_coeff.size != 0, "Beta spline coefficient array (t_coeff) is empty"
            my_spline, ctr_pts, ctr_coeffs = _bspline(b_coeff)
            self.my_spline = my_spline(self.s)
        else:
            def constant_function(input):
                """
                Return array of ones same as the size of the input array. This
                function is called when Beta spline function is not used.

                Parameters
                ----------
                input

                Returns
                -------

                """
                return np.ones(input.shape)

            self.my_spline = constant_function(self.s)

    def apply_torques(self, rod: RodType, time: np.float64 = 0.0):
        self.compute_lifting_torques(
            time,
            self.my_spline,
            self.s,
            self.angular_frequency,
            self.wave_number,
            self.Amplitude,
            self.lamBDA,
            self.phase_shift,
            self.ramp_up_time,
            self.direction,
            rod.director_collection,
            rod.external_torques,
        )

    def compute_lifting_torques(
        time,
        my_spline,
        s,
        angular_frequency,
        wave_number,  # k pf muscle torque
        Amplitude,
        lamBDA,
        phase_shift,
        ramp_up_time,
        direction,
        director_collection,
        external_torques,
    ):
        # Ramp up the muscle torque
        factor = min(1.0, time / ramp_up_time)
        
        # From the node 1 to node nelem-1
        # Magnitude of the torque. Am = beta(s) * sin(2pi*t/T + 2pi*s/lambda + phi)
        # There is an inconsistency with paper and Elastica cpp implementation. In paper sign in
        # front of wave number is positive, in Elastica cpp it is negative.
        # torque_mag = (
        #    factor
        #    * my_spline
        #    * np.sin(angular_frequency * time - wave_number * s + phase_shift)
        # )
        kt = wave_number * lamBDA
        torque_mag = max(0, (
            factor
            * my_spline
            * Amplitude * np.cos(angular_frequency * time + kt * s + phase_shift) + 1
        ))
        
        # Head and tail of the snake is opposite compared to elastica cpp. We need to iterate torque_mag
        # from last to first element.
        torque = _batch_product_i_k_to_ik(direction, torque_mag[::-1])
        inplace_addition(
            external_torques[..., 1:],
            _batch_matvec(director_collection, torque)[..., 1:],
        )
        inplace_substraction(
            external_torques[..., :-1],
            _batch_matvec(director_collection[..., :-1], torque[..., 1:]),
        )

AkashVardhan7, 2023-10-17:
I have retained everything which was present in the class MuscleTorques with the addition of variables lamBDA and Amplitude... which represent those two parameters from the paper

AkashVardhan7, 2023-10-17:
Okay, so I got the simulation to run with the liftingtorques class after fiddling around with the indentation some more and using np.maximum to apply the liftingTorques rather than just max., but there is no snake in the final simulation and all I see is a blank.. when I comment out the (liftingTorques) part in the simulation and run it only with the muscle torques, I see the snake moving along a straight line as before.. clearly I am doing something wrong...

AkashVardhan7, 2023-10-17:
I put my code in a repo and shared it with you, it'd be helpful if you could try running it on your end to see what I am saying.

Ali-7800, 2023-10-18:
@AkashVardhan7 I think the problem is in `torque_mag = np.maximum(0,(factor* my_spline* Amplitude* np.cos(angular_frequency * time + kt * s + phase_shift)+ 1 ),)`, I think it should be ```torque_mag = np.maximum(0,(factor* my_spline* (Amplitude* np.cos(angular_frequency * time + kt * s + phase_shift)+ 1) ),)``` instead.

AkashVardhan7, 2023-10-18:
[image]
Ali I changed it to what you suggested, yet the problem persists. :(

armantekinalp, 2023-10-20:
@xzhan139 can you also take a look at this issue.

AkashVardhan7, 2023-10-21:
I shared the repo with @xzhan139

AkashVardhan7, 2023-10-23:
Any further pointers on what I should be doing to make this work? I'd want to capture all the behaviors you reported in the manuscript before I can reliably start using this sim for my project...

xzhan139, 2023-10-23:
Hi can you share your repo again? I didn't get any invitation yet

AkashVardhan7, 2023-10-23:
https://github.com/AkashVardhan7/snake_diffraction_Elastica.git

Here's the link to the repo (I made it public so you should be able to access the code... I made modifications to externalforces.py and __init__.py as you can follow from the thread.. here's a snapshot of the invitation I sent you
[image]

xzhan139, 2023-10-23:
Thanks for sharing! Sorry I missed your invitation on Saturday.
A couple of things  - 
1. In your lifting force
[image]
If I didn't misunderstand, "direction = direction"  is twist direction, not lifting direction. Adding twist in general is very tricky as you have to be very careful with the torque amplitude and timestep, otherwise your simulation diverge very quickly. I guess that is why you don't see the snake, it exploded.  Lifting is instead in binormal direction.
2. When you are adding new force, it's generally recommended to start with a very small force value, just to test the algorithm implementation itself. Big force itself can make simulation unstable, so it's harder to figure out where the problem is.
3. An alternative way of doing this - lifting wave and lateral wave are essential the same thing, just that the normal direction of the wave is different. You can eventually merge the two forcing class into one. But of course, this is after resolving the current issue you have.
Please let me know if these help.

AkashVardhan7, 2023-10-23:
[image]
These are the lifting torque parameters with which I am running my simulation... lateral_direction is the same binormal direction vector pointing along x axis with unit vector [1.0,0.0,0.0]. The video shows that the snake behaves chaotically after slipping, which is indicating that there is an unbalance in the net forces somewhere., while I am running it with a time step of 1e-4.

https://github.com/GazzolaLab/PyElastica/assets/48886152/24f33303-cd8b-468c-8e0e-9c61ec7cc7e8

xzhan139, 2023-10-24:
Hi, this looks to me like a numerical unstability due to your parameter settings. As I can see from your code there are few parameters that are quite different from what we usually run with, for example you have a period = 15, which makes the snake run very slow.
We will push a case in PyElastica soon, which will include functions for lifting wave etc. The parameters of the snake will be slightly different, but you can use it as a reference for your implementation. Thank you.

AkashVardhan7, 2023-10-24:
Thanks please let me know when you have done so, and I can modify the code accordingly..When I changed the period to 2 second which is what is used in the paper SI and the time step to 8e-6 again the value used from the period, I still find that
[image]
 the numerical issue persists,

xzhan139, 2023-10-24:
couple of other things to try -
1. Chang the number of rod elements from 50 to 25. This will make each element heavier, will in general enhance numerical stability. 
2. Increasing your damping constant to 1e-1. 
3. Initialize your snake in the air, remove gravity, remove lateral wave, and plot the snake sideview video. You should see the snake floating and doing lifting wave. If this is not stable, then it is something in your lifting wave forcing itself.

AkashVardhan7, 2023-10-24:
The damping and number of rods helped and made the simulation more stable, I am observing sidewinding like behavior but the trajectory still seems a little funky in terms of the dynamics. I will try turning off the lateral wave and plotting just the side view with snake initialized in air.

https://github.com/GazzolaLab/PyElastica/assets/48886152/7529ec8a-2e46-42b9-b1d1-806a4f823b2d

AkashVardhan7, 2023-10-24:
I think it's the phase offset and the wavelength of the lifting wave which is giving the snake this waveform, I am using the same wavelength as the lateral wave and a phase shift of 0.25, which is chosen from the region in the parameter space supposed to generate straight side winding for lambda = 1.

AkashVardhan7, 2023-10-24:
Here's the lifting wave without the lateral wave and gravity.
[image]

AkashVardhan7, 2023-11-14:
This is the best side-winding I got 
[image]
When are you going to push your version of the code with sidewinding on github?

Ali-7800, 2023-11-15:
Hi @AkashVardhan7,

Sorry for the delay I had to resolve some issues, I just opened the pull request with the new continuum snake with lifting wave example, it is in review now.

Ali-7800, 2023-12-05:
Hi @AkashVardhan7,

The continuum snake with lifting wave case has been add, you can check it out in the update-0.3.2 branch

armantekinalp, 2023-12-05:
Since we have merged PR, I am closing this issue. If you have questions, please reopen again.

closed 2023-12-05 by armantekinalp