GazzolaLab/PyElastica #308: Instant controler and simulation

Junang-Wang, 2023-10-17:
Hi there, 

I am a beingger of PyElastica, does anyone know if PyElastica support real time control applications? For example, one can modify externel force from a keyboard or a controler, then PyElastica simulate real time configuration. And does anyone know how to implement it?

I appreciate for the help advance.

armantekinalp, 2023-10-17:
Hi @Junang-Wang 

Although we have hooked up PyElastica with RL and model based controllers we haven't haven't tried controlling PyElastica rods with keyboard. Depending on the properties of the rod (soft rods) you can achieve real time simulations. Then you can design the interface with a keyboard controller to achieve your goal.

Ali-7800, 2023-10-17:
HI @Junang-Wang,
Here's a code snippet to help you with controlling the rod with your keyboard. I modified the AxialStretchingCase example to control the stretching force using the keyboard:

```
# ea.integrate(timestepper, stretch_sim, final_time, total_steps)
# modify the loop in ea.integrate to work with keyboard press

from elastica.timestepper import extend_stepper_interface
from tqdm import tqdm
import keyboard

do_step, stages_and_updates = extend_stepper_interface(timestepper, stretch_sim)

dt = np.float64(float(final_time) / total_steps)
time = 0

for i in tqdm(range(total_steps)):
    time = do_step(timestepper, stages_and_updates, stretch_sim, time, dt)
    if keyboard.is_pressed('a'):
        print("a is pressed")
        end_force[:] += dt*np.array([1.0,0.0,0.0]) #increase force when a is pressed (scaled by dt)
    else:
        temp = end_force.copy() -  dt*np.array([1.0,0.0,0.0]) #decrease force when a is not pressed (scaled by dt)
        if temp[0] > 0:
            end_force[:] = temp #if greater than zero then use temp force
        else:
            end_force[:] = np.array([0.0,0.0,0.0]) #if less than zero then use zero force
```

https://github.com/GazzolaLab/PyElastica/assets/47090295/52cbddba-19f9-4898-9bfb-9395c11c2808

Junang-Wang, 2023-10-18:
Hi @Ali-7800 @armantekinalp, thank you so much for your reply. Ali, your code snippet is very heuristic for me. Do you guys know where I can find the tutorial or document of PyElastica RL and model based controllers modules? I didn't find the information of control/Robotic modules, for example extend_stepper_interface in Ali's code, in the PyElastica doc page  [https://docs.cosseratrods.org/en/latest/index.html](url).

armantekinalp, 2023-10-20:
I am closing this issue seems it is resolved

closed 2023-10-20 by armantekinalp