GazzolaLab/PyElastica #315: Adding force and torque at same time

AaronRapopot, 2023-10-30:
Hello. I applied a torque to bend my rod. Now, I also want to apply a force in the tangential direction along the rod at the free endpoint. Is there a good method to achieve this? Like this:

[image]

ilianasiriziba, 2023-11-02:
Hi @AaronRapopot,

Yes, for this aim you can add EndpointForces on the rod. Just use the tangent argument of the rod to define the force direction. You need to make the force direction equal to the tangent of the last element.

AaronRapopot, 2023-11-03:
Hi @ilianasiriziba 
The direction of the Endpointforce should change continuously as the rod bending. Can I obtain the tangential direction of the rod's endpoint in real-time through a callback?Perhaps it needs to be calculated step by step?

For example:
for step in range(initial_step, final_step + 1):
    tangent_direction = get_tangent_direction(step - 1) #Get the position information from the previous step through a callback
    force_vector = tangent_direction * endpointforce
    apply_force_to_rod(force_vector)

Not actual code, just to explain the logic. Thanks for your help!

ilianasiriziba, 2023-11-06:
Hi @AaronRapopot,

If you make the force direction equal to rod_name.tangents of the last element it will be updated step by step.

bhosale2, 2024-02-01:
@AaronRapopot Has your issue been resolved? Please let us know, or else we shall close this issue due the inactivity in a few days.

bhosale2, 2024-02-13:
@ilianasiriziba and @armantekinalp closing this issue due to inactivity, as mentioned above. Feel free to open if deemed necessary.

closed 2024-02-13 by bhosale2