GazzolaLab/PyElastica #333: Avoid having randomized unittest

skim0119, 2024-02-20:
https://github.com/GazzolaLab/PyElastica/blob/d12107a2d3ad7228a2893ad55de5da89a9bab87b/tests/test_math/test_timestepper.py#L401-L408

vikas-chaudhary-2802, 2024-02-23:
Hi, I love to work on this issue, Can you please assign me ??

skim0119, 2024-02-23:
@vikas-chaudhary-2802 I included some more explanation above.

vikas-chaudhary-2802, 2024-02-27:
@skim0119 I have made few changes and i wanted to test whether those are working. Do i need to build the project and then run the test using 'make test' command?
If yes, how can i build it?

skim0119, 2024-02-28:
@vikas-chaudhary-2802 Everything can be installed using `pip` or `poetry`.
- [install](https://github.com/GazzolaLab/PyElastica/blob/d68887cb72649c8b24227d57f68bca93850a7bf6/Makefile#L16-L17)
- [test](https://github.com/GazzolaLab/PyElastica/blob/d68887cb72649c8b24227d57f68bca93850a7bf6/Makefile#L65-L66)

vikas-chaudhary-2802, 2024-02-28:
@skim0119  whether I need to build after I make the changes or not ?

vikas-chaudhary-2802, 2024-03-05:
i added logging line to check the datatype but the logs are not getting printed on the shell when i run - make tests....could you please suggest any other way to debug ?

skim0119, 2024-03-06:
@vikas-chaudhary-2802 You can directly run `pytest -s` on repository to capture `stdout/stderr`

vikas-chaudhary-2802, 2024-03-18:
@skim0119 @bhosale2 I'm facing difficulties with certain test cases and despite my attempts, I haven't been able to resolve them. Could you please provide assistance with these specific issues?

vikas-chaudhary-2802, 2024-03-18:
@skim0119 @bhosale2  I'm facing difficulties with certain test cases and despite my attempts, I haven't been able to resolve them. Could you please provide assistance with these specific issues?
[image]

skim0119, 2024-03-18:
@vikas-chaudhary-2802 Could you commit your code in your `fork` that reproduces this error?

vikas-chaudhary-2802, 2024-03-19:
@skim0119 yeah, I am able to commit 
[image]

vikas-chaudhary-2802, 2024-03-19:
@skim0119 @bhosale2  I Create a PR,  Can you please suggest me the changes that i have to do ...so that the failed test cases will resolve and then i will commit the changes again

skim0119, 2024-03-23:
@vikas-chaudhary-2802 I went over your code. The directors passed to the timestepper need to be orthogonal. I think that is why the analytical solution is breaking. Please correct the directors and try to run the test again.

Also, try to format the script before commiting. You can run `make format-codestyle`.

vikas-chaudhary-2802, 2024-03-24:
Hi, @skim0119 Thank You for your suggestion. I will make the changes as soon as possible

vikas-chaudhary-2802, 2024-03-26:
Hi, @skim0119 @bhosale2 Now the directors passed to the timestepper is orthogonal but still the same test cases failed as i mentioned above i will attached the changes
[image]

[image]

skim0119, 2024-03-26:
@vikas-chaudhary-2802 I would appreciate it if you could attach the actual code, or link the forked script to re-run. I can't help out just looking at the image file. Same for the error message.

Few comments:
- I am not sure how you are initializing `random_directors`. Please re-write them readable and clean. The final dimension needs to be `3x3xN` where N is the number of test cases.
- Don't use QR decomposition to orthogonalize them. The whole point of this issue and to avoiding randomized unit-test is to test if the code can handle largely expected cases (and corner cases if any). Your approach is, in the end, same as the original implementation: don't know why the test passes or fails.

skim0119, 2025-08-09:
Fixed in update-0.3.3

closed 2025-08-09 by skim0119