GazzolaLab/PyElastica #343: Examples not running due to modules not being found

jam474cwru, 2024-03-10:
Description: Hello. I am new to the use of PyElastica and haven't coded in some time. I am trying to learn this open source software and run the examples to understand and learn from both the binder and examples folders. However, no matter how many times I try to address the errors, I always am told the beginning modules are not found with the ModuleNotFound error. It is quite frustrating and confusing especially since I would expect them to be within the software already. I am particularly just wanting to start by learning the binder examples and the timoshenkobeamcase and muscularsnake examples. 

To reproduce, all I have simply done is copy the exact code from github and cloned a repository in Visual Studio Code. 

Expected Behavior: ModuleNotFound 

[image]

[image]

jam474cwru, 2024-03-10:
@armantekinalp

armantekinalp, 2024-03-11:
Based on the first image you shared check if you have installed PyElastica correctly. Make sure to create a conda environment with python 3.10. Then install pyelastica doing `pip install pyelastica` . I suggest also install example dependencies `pip install "pyelastica[examples,docs]" `. 

Based on second image python cannot find the paths, I suggest appending paths.

```  python 

import sys
sys.path.append("/")
sys.path.append("../")
sys.path.append("../../")

```

jam474cwru, 2024-03-11:
@armantekinalp Thank you for the response, Arman. I installed and created the conda environment with python 3.10 and ran those couple initial installations of pyelastica. I even did the second recommendation in relation to the path, but did not have any luck. I copied that code in a separate file as well and even tried in front of the timoshenko example. Any other recommendations?

[image]

jam474cwru, 2024-03-11:
@armantekinalp I have even tried:

import sys
sys.path.append(r"c:\Users\jmegu\PyElestica\PyElastica\examples\TimoshenkoBeamCase\timoshenko.py")

thinking that this is the path I must fix, but nothing has worked.

Ali-7800, 2024-03-11:
[image]

If you are using VScode, try enabling this option in the settings. You can find it by searching for python terminal in the settings.

jam474cwru, 2024-03-11:
[image]

Unfortunately, the same outcome after I enabled that in settings. I even created a new window and folder and cloned the code from the repository all over again. I installed pyelastica and the initial steps again, but don't understand why this is happening.

jam474cwru, 2024-03-11:
@Ali-7800

jam474cwru, 2024-03-11:
I do receive this in the bottom right corner when I close out of VS code and reopen:

[image]

Ali-7800, 2024-03-11:
Please keep the 
```  python 

import sys
sys.path.append("/")
sys.path.append("../")
sys.path.append("../../")

```

jam474cwru, 2024-03-17:
I believe this worked in running the initial codes. Thank you. However, what is the requirements to run a movie? In the muscular snake example, I ran the code for multiple days and by the end it was not able to do anything as it had a runtime error and issue with a needed requirement for a software for any sort of movie. Is there an extension that is needed for these final plots to work? 

Unfortunately I killed the terminal so I couldn't post a picture as it ran the simulation again automatically.

armantekinalp, 2024-03-17:
do this `pip install "pyelastica[examples,docs]"` to install example dependencies. Before running the whole simulation change final time to something small and make sure you can generate the videos. Then run the whole simulation.

jam474cwru, 2024-03-18:
Upon trying this, I get this error:

[image]

Ali-7800, 2024-03-18:
Do `conda install ffmpeg`

jam474cwru, 2024-03-19:
[image]

Seems there is an OS error of not having a "seaborn-whitegrid" file. Are these common errors that are often experienced?

armantekinalp, 2024-03-19:
Can you check you matplotlib version? Seems like `seaborn-whitegrid` is depreciated. You can install matplotlib version `3.7.5  ` or you can checkout the branch `update-0.3.2` , I fixed it there.

jam474cwru, 2024-03-19:
I was able to install that version and it worked... super cool to be able to visualize it. What is the best way to save work? I am struggling downloading the mp4 videos and saving.

armantekinalp, 2024-03-20:
what do you mean? mp4 videos should be saved in the same directory as the snake file.

jam474cwru, 2024-04-04:
I was able to get the videos. Thank you. 

I particularly would like to inquire about the muscular snake example. What is the difference between nu and poissons ratio when defining the bodies? what about n_elem_body? How are the force amplitudes determined and for which muscles? 

I really appreciate your help through these questions.

skim0119, 2024-04-04:
@jam474cwru I am closing this issue since this particular problem seems to be resolved. If you have further questions, please open another issue/discussion.

I believe the answer for the questions that you asked can be found in the tutorials, I would recommend to check [tutorial: theory](https://www.cosseratrods.org/cosserat_rods/theory/) and [tutorial: numerics](https://www.cosseratrods.org/cosserat_rods/numerics/).

closed 2024-04-04 by skim0119