GazzolaLab/PyElastica #411: What does director_collection mean specifically？

DISCORDzzz, 2024-07-08:
Hi! I am currently doing some development based on `Elastica+RL`, but I found that `Elastica+RL` has stopped updating, so I am looking for help here.
I want to know what `self.shearable_rod.director_collection` means? I noticed that each element in it is a `3*3` array, which seems to be used to indicate the direction of the rod. But I can't understand its specific meaning. For example, I want to be able to approach a point on the rod vertically along a certain direction. I think this seems to be obtained through the information provided by `self.shearable_rod.director_collection[..., x]`, where **x** is a certain information point on the rod. But I currently can't understand the meaning of the specific elements of `self.shearable_rod.director_collection`. Can you help me solve it?
Thanks for your contribution!

armantekinalp, 2024-07-08:
Hi @DISCORDzzz 

The `self.shearable_rod.director_collection` represents the frames attached to each element of the rod, organized in a size of 3 × 3 × n_elem. For a more detailed explanation, please refer to this [paper](https://mattia-lab.com/wp-content/uploads/2018/06/Gazzola_RSOS_2018.pdf), where the concept of directors is illustrated using 
𝑄 throughout the text.

armantekinalp, 2024-07-08:
Also I am converting this to a discussion since it is a question.