GazzolaLab/PyElastica #420: Computatuion of v_i (t+ 0.5 *delta t) and omega_i (t+ 0.5 *delta t).

Srishti-Siddharth-iitb, 2024-07-24:
I am trying to simulate a single Cosserat rod in MATLAB using your codes and the numerical scheme discussed in the following 2018 paper: https://doi.org/10.1098/rsos.171628. According to equations (C5) and (C6) of the paper (page 25), the position and orientation at (t+delta t) can be computed using  v_i (t+ 0.5 * delta t) and omega_i (t+ 0.5 *delta t). Could you please confirm whether these velocities have been computed using the following formulae in your Python code?

v_i (t+ 0.5 *delta t) = v_i (t) +0.5 * delta t* dv_dt(t+0.5* delta t),

 and
 
omega_i (t+ 0.5 *delta t) = omega_i (t) +0.5 * delta t* domega_dt(t+0.5 *delta t),

armantekinalp, 2024-07-24:
[image]

There is a typo there. You can use the above equations.

sy-cui, 2024-08-21:
In our code we use $\omega$ at time $t$, although I do think you can use the formulae you proposed. For the time-stepping method we use (Verlet integration) to be applicable, technically $d\omega/dt$ should not be a function of $\omega$. However, the unsteady dilatation and Lagrangian transport terms are rarely important, and the overall algorithm is first-order accurate in time nonetheless.

armantekinalp, 2024-08-21:
I am converting this to a discussion since it more about the methods, than code issue.